karateboy/SensorManager #3: 衰減月報：待調整部分

haopingchung, 2021-05-27:
- [ ] 1. excel範本修正(新增全區域小時值平均)-alan

- [ ] 2. 全區域bias公式修正(將同時間之感測器濃度平均後再計算bias、若有缺值或0則不納入計算)

- [ ] 3. EPA測站若原始資料為空格，資料庫會帶入0問題修正

- [ ] 4. 完成全區域濃度趨勢圖功能(200及210分開、x軸時間長度為一年)

karateboy, 2021-05-29:
已修正

haopingchung, 2021-05-30:
- [ ] 1. 發現範例excel中工作表1~3中的bias公式中多了*100，已在0527版中修正，再請協助更改

- [ ] 2. 下載的excel工作表1，第一列及第二列右側還留有範本中的內容

- [ ] 3. EPA無效值(空格)帶入0的問題仍在(3/11~3/12)

- [ ] 4. 建議將趨勢圖直接放入工作表5及6

- [ ] 5. 工作表4中R2有負數似乎不合理?

karateboy, 2021-05-31:
1. 已修正
2. 已修正
3. 已經重新匯入3月資料, 修正
4. Excel圖表無法直接參考目前的工作表
5. 已經修正錯誤公式

haopingchung, 2021-06-01:
- [ ] 1. excel報表中bias公式*100仍在(如圖)
[image]

- [ ] 2. 工作表名稱修正
          a.兩個3.
          b.200型趨勢圖>200、210型趨勢圖

karateboy, 2021-06-01:
已經修正

closed 2021-06-01 by haopingchung