karateboy/SensorManager #8: 6/3會議討論內容

haopingchung, 2021-06-03:
一、月報
- [ ] 1. 工作表2 C752需移除*100
- [ ] 2. 最近會上傳新excel範本至onedrive
- [ ] 3. 離群分析範例excel

二、異常查詢
- [ ] 1.  地圖filter新增"電源異常"(同步新增於即時狀況)
           以時間判定，2000~0700內若回傳數值為1(代表使用內部電池)，則代表異常
- [ ] 2. 於數據查詢新增"異常資訊查詢"，內容和即時狀況相同(需可選擇日期)，僅7:00am執行一次，並將結果匯入資料庫

三、資料查詢
- [ ] 1. 資料庫小時值算法修改(小時平均值)
- [ ] 2. 系統管理新增"測點基本資料"查詢功能(即各區域感測器資料表)，若於平台修改內容，相關聯資料也會同步變動

四、美編
- [ ] 1. 濃度圖縮小(盈如製作中)
- [ ] 2. 地圖上EPA、感測器濃度圖框更換(大圓圈為EPA)
[image]
- [ ] 3. 儀錶板 即時資訊/SAQ-200及210，移除/
- [ ] 4. 地圖filter美化(自由發揮)

五、下月討論事項
1. 異常狀況通報line & email
2. 直接於平台調整感測器接收測項(PM10、H2S、噪音)

karateboy, 2021-06-03:
將原本issue 拆分開來, 方便追蹤

haopingchung, 2021-06-04:
2. 以上傳excel(0604)

haopingchung, 2021-06-04:
3. 離群分析初稿
[基隆離群分析初稿(0604).xlsx](https://github.com/karateboy/SensorManager/files/6597643/0604.xlsx)
說明(以K9SE01為例)：選擇地區與月份後，網頁上呈現盒鬚圖，但可以下載原始excel查看

karateboy, 2021-06-05:
> 1. 已上傳excel(0604)
>    [基隆衰減分析excel(0604).xlsx](https://github.com/karateboy/SensorManager/files/6597499/excel.0604.xlsx)
已經更新, 原本整群平均與基隆測站的趨勢圖你刪掉了?

karateboy, 2021-06-06:
離群分析已經更新, 但只有第一個群組有圖?

haopingchung, 2021-06-07:
2. 200及210趨勢圖漏掉了，已補上
[基隆衰減分析excel(0607).xlsx](https://github.com/karateboy/SensorManager/files/6605520/excel.0607.xlsx)

haopingchung, 2021-06-07:
3. 已更新檔案
[基隆離群分析(0607).xlsx](https://github.com/karateboy/SensorManager/files/6605911/0607.xlsx)

haopingchung, 2021-06-07:
3. 已更新(補上其他群組的圖)
[基隆離群分析(0607).xlsx](https://github.com/karateboy/SensorManager/files/6606058/0607.xlsx)

另外請問，盒鬚圖是否有辦法直接呈現於網頁上?

karateboy, 2021-06-08:
盒鬚圖可以呈現在網頁上....要分群組?

haopingchung, 2021-06-08:
選擇縣市後可以選群組(可多選or全選)

closed 2021-06-17 by haopingchung