karateboy/SensorManager #16: 6/11：報表查詢

haopingchung, 2021-06-11:


haopingchung, 2021-06-18:
6/18 離群分析報表調整
[基隆離群分析(0618).xlsx](https://github.com/karateboy/SensorManager/files/6676257/0618.xlsx)

karateboy, 2021-06-18:
已經完成

haopingchung, 2021-06-21:
離群分析
- [ ] 1. 網頁上可否於最左側新增整群平均之盒鬚圖(如excel中B欄)，並用該欄中位數畫紅線
- [ ] 2. 以上述離群條件判斷，若該感測器離群，圖面上有醒目提示(盒鬚圖顏色改變or特殊標記)
- [ ] 3. 部分圖表"中位數平均"會被圖擋住
- [ ] 4. 盒鬚圖中最大&最小值改為上限&下限，公式如下：
[image]
excel已修改：
[基隆離群分析(0621).xlsx](https://github.com/karateboy/SensorManager/files/6683735/0621.xlsx)
- [ ] 5. 網頁上群組可否複選(一次呈現多張圖)

karateboy, 2021-06-22:
5. 有空再弄...

haopingchung, 2021-06-23:
- [ ] 1. 平台上最大最小值計算方式與excel不同，平台目前是Q3+1.5(Q3-**median**)及Q1-1.5(Q3-**median**)
- [ ] 2. 小數點後固定2位
[image]

- [ ] 3. 如下圖，1233及1272應為離群但沒有顯示
[image]

- [ ] 4. 若設備離群可否以紅字標示較清楚

karateboy, 2021-06-24:
是Excel的內容錯了. 
[image]

https://zh.wikipedia.org/wiki/%E5%9B%9B%E5%88%86%E4%BD%8D%E8%B7%9D

karateboy, 2021-06-24:
> * [x]  1. 平台上最大最小值計算方式與excel不同，平台目前是Q3+1.5(Q3-**median**)及Q1-1.5(Q3-**median**)
> * [x]  2. 小數點後固定2位
>   [image]
> * [x]  3. 如下圖，1233及1272應為離群但沒有顯示
>   [image]
> * [ ]  4. 若設備離群可否以紅字標示較清楚

離群但沒有顯示  是因為公式不同

haopingchung, 2021-06-25:
[image]
QD是四分位差，但上下限是以四分位距(IQR)來計算的喔@@

karateboy, 2021-06-29:
已經支援多重群組選擇
另可選擇是否顯示farout

closed 2021-06-30 by haopingchung