karateboy/SensorManager #20: 6/18：美編

haopingchung, 2021-06-18:
- [ ] 1. 新icon如ppt
[sensor_icon.pptx](https://github.com/karateboy/SensorManager/files/6675638/sensor_icon.pptx)

- [ ] 2.  儀錶板統計資料調整
[感測器平台儀表板設計.xlsx](https://github.com/karateboy/SensorManager/files/6675642/default.xlsx)

- [ ] 3. 地圖_圖層選擇：移除白底、圖層選擇字樣，大致效果如下(紅框部分)
[image]

- [ ] 4. 儀錶板200、210的更新時間位置調整
[image]

- [ ] 5. 夜間模式下時間選擇改為黑底

haopingchung, 2021-06-23:
- [ ] 1. 電力異常也麻煩先用ppt裡面的icon
- [ ] 2. 通訊中斷的圖本身應該是去背的，不過地圖上呈現會有白底可否調整
- [ ] 3. 由於excel匯出匯入的icon原本有白底，故換成新icon，並移除目前綠底(包含即時狀況裡的匯出)
[image]
[image]
- [ ] 4. 環保署&感測器勾選 字體加大1倍

karateboy, 2021-06-23:
[sensor_icon.pptx](https://github.com/karateboy/SensorManager/files/6698707/sensor_icon.pptx)
這是目前Icon繪製的ppt

haopingchung, 2021-06-24:
[icon0624.pptx](https://github.com/karateboy/SensorManager/files/6705831/icon0624.pptx)

closed 2021-06-30 by haopingchung