karateboy/SensorManager #22: 6/18：其他

haopingchung, 2021-06-18:
- [ ] 1. 感測器之接收測項增加(目前僅2.5、濕度)，增加PM10、溫度、TVOC、O3、AC/DC、H2S、NH4、NO2

- [ ] 2. 歷史資料查詢、歷史趨勢圖：測項選擇清單僅保留有接收的測項

- [ ] 3. 測點管理之資料表補上經緯度及群組

karateboy, 2021-06-21:
目前, SAQ200只有測量
 { "sensor": "co", "value": "NA", "unit": "ppb" },
      { "sensor": "o3", "value": "NA", "unit": "ppb" },
      { "sensor": "noise", "value": "NA", "unit": "dB" },
      { "sensor": "voc", "value": 24, "unit": "" },
      { "sensor": "pm2_5", "value": 12, "unit": "繕g/m3" },
      { "sensor": "pm1", "value": 7, "unit": "繕g/m3" },
      { "sensor": "pm10", "value": 15, "unit": "繕g/m3" },
      { "sensor": "no2", "value": "NA", "unit": "ppb" },
      { "sensor": "humidity", "value": 92.33, "unit": "%" },
      { "sensor": "temperature", "value": 29.05, "unit": "??" },
      { "sensor": "humidity_main", "value": 68.48, "unit": "%" },
      { "sensor": "temperature_main", "value": 34.24, "unit": "??" },
      { "sensor": "volt", "value": 36531, "unit": "v" },
      { "sensor": "ampere", "value": 63745, "unit": "mA" },
      { "sensor": "devstat", "value": 0, "unit": "" }
SAQ210
    "data": [
      { "sensor": "voc", "value": 158, "unit": "" },
      { "sensor": "pm2_5", "value": 5.89, "unit": "繕g/m3" },
      { "sensor": "pm1", "value": 2, "unit": "繕g/m3" },
      { "sensor": "pm10", "value": 2, "unit": "繕g/m3" },
      { "sensor": "humidity", "value": 76.26, "unit": "%" },
      { "sensor": "temperature", "value": 31.32, "unit": "??" },
      { "sensor": "humidity_main", "value": 53.61, "unit": "%" },
      { "sensor": "temperature_main", "value": 38.28, "unit": "??" },
      { "sensor": "volt", "value": 32775, "unit": "v" },
      { "sensor": "ampere", "value": 18876, "unit": "mA" },
      { "sensor": "battery_current_1", "value": 1604, "unit": "" },
      { "sensor": "battery_voltage_1", "value": 8235, "unit": "" },
      { "sensor": "rssi", "value": -51, "unit": "" }
    ]

haopingchung, 2021-06-23:
- [ ] 1. 歷史資料查詢/趨勢圖的測項選擇有辦法選擇AC/DC查看嗎

closed 2021-06-23 by haopingchung