karateboy/SensorManager #30: 7/13：異常通報

haopingchung, 2021-07-13:
- [ ] 1. 目前每晚8點傳送前一日的異常清單，改為每晚八點半傳送當日的異常清單

karateboy, 2021-07-14:
目前只通報電力異常, 其他的定值和完整率不足90% 需要增加嗎?
另外上次盈如不是說要讓使用者選擇縣市?

haopingchung, 2021-07-15:
- [ ] EMAIL通報縣市選擇(基隆、屏東、宜蘭)功能

定值跟完整率我認為暫時還不需要

karateboy, 2021-07-21:
已經增加縣市選擇

haopingchung, 2021-07-22:
已設定 今天晚上會看一下

haopingchung, 2021-07-25:
這2天都沒有收到mail的樣子

karateboy, 2021-07-25:
剛剛看log  就是網路怪怪的....google是說有可能是防毒軟體或防火牆造成的
2021-07-23 20:30:50,474 [ERROR] from application in ForkJoinPool-2-worker-7 - Failed to send email
org.apache.commons.mail.EmailException: Sending the email to the following server failed : smtp-relay.gmail.com:587
Caused by: javax.mail.MessagingException: Could not convert socket to TLS
Caused by: javax.net.ssl.SSLException: readHandshakeRecord
Caused by: java.net.SocketException: Software caused connection abort: socket write error

karateboy, 2021-07-25:
今晚再看看能不能順利寄出  我軟體都先不更新

karateboy, 2021-07-25:
今天八點半又出現同樣的問題...網路突然斷線  連不到gmail發信...
2021-07-25 20:29:58,882 [ERROR] from application in ForkJoinPool-2-worker-27 - Failed to send email
org.apache.commons.mail.EmailException: Sending the email to the following server failed : smtp-relay.gmail.com:587
	at org.apache.commons.mail.Email.sendMimeMessage(Email.java:1421)
	at org.apache.commons.mail.Email.send(Email.java:1448)
	at play.api.libs.mailer.SMTPMailer$$anon$2.send(MailerPlugin.scala:100)
	at play.api.libs.mailer.CommonsMailer.send(MailerPlugin.scala:130)
	at play.api.libs.mailer.SMTPMailer.send(MailerPlugin.scala:110)
	at play.api.libs.mailer.SMTPDynamicMailer.send(MailerPlugin.scala:117)
	at models.ErrorReportOp$$anonfun$sendEmail$1$$anonfun$apply$1.apply(ErrorReport.scala:163)
	at models.ErrorReportOp$$anonfun$sendEmail$1$$anonfun$apply$1.apply(ErrorReport.scala:153)
	at scala.collection.immutable.List.foreach(List.scala:392)
	at models.ErrorReportOp$$anonfun$sendEmail$1.apply(ErrorReport.scala:153)
	at models.ErrorReportOp$$anonfun$sendEmail$1.apply(ErrorReport.scala:140)
	at scala.util.Success$$anonfun$map$1.apply(Try.scala:237)
	at scala.util.Try$.apply(Try.scala:192)
	at scala.util.Success.map(Try.scala:237)
	at scala.concurrent.Future$$anonfun$map$1.apply(Future.scala:237)
	at scala.concurrent.Future$$anonfun$map$1.apply(Future.scala:237)
	at scala.concurrent.impl.CallbackRunnable.run(Promise.scala:36)
	at scala.concurrent.impl.ExecutionContextImpl$AdaptedForkJoinTask.exec(ExecutionContextImpl.scala:121)
	at scala.concurrent.forkjoin.ForkJoinTask.doExec(ForkJoinTask.java:260)
	at scala.concurrent.forkjoin.ForkJoinPool$WorkQueue.runTask(ForkJoinPool.java:1339)
	at scala.concurrent.forkjoin.ForkJoinPool.runWorker(ForkJoinPool.java:1979)
	at scala.concurrent.forkjoin.ForkJoinWorkerThread.run(ForkJoinWorkerThread.java:107)
Caused by: com.sun.mail.util.MailConnectException: Couldn't connect to host, port: smtp-relay.gmail.com, 587; timeout 60000

haopingchung, 2021-07-26:
會跟上次更新縣市選擇有關係嗎?剛好更新完後就沒收到了

closed 2021-08-03 by haopingchung