karateboy/SensorManager #50: 7/21：10000新群組問題回報

haopingchung, 2021-07-21:
- [ ] 1. 設定新群組若測項只選擇PM2.5，儀表板會顯示無資料
[image]
[image]

要將四個測項都選擇才會顯示
[image]
[image]

karateboy, 2021-07-21:
我創了一個環佑的使用者看起來一切正常
[image]

haopingchung, 2021-07-22:
我早上試又可以了XD

closed 2021-07-22 by haopingchung