karateboy/SensorManager #53: 7/30討論內容：主畫面(調整)

haopingchung, 2021-08-03:
- [ ] 1. 監測地圖字樣大小固定
- [ ] 2. 監測地圖時間固定為一行
- [ ] 3. 監測地圖異常狀況僅保留定值、通訊中斷、充電異常
- [ ] 4. 異常狀況定義調整：定值(0700~0730 ; 使用者可自行定義)、通訊中斷(0705~0710)、充電異常(2000~2010)
- [ ] 5. 宜蘭完整率<90%數量確認

karateboy, 2021-08-05:
已經修正

haopingchung, 2021-08-09:
- [ ] 1. 首頁檢核資料修改(括號時間部分)+移除完整率
定值(0700~0730)
通訊中斷(0705~0710)
充電異常(2000~2010)
[image]

- [ ] 2. 定值判定時間是否可讓使用者自行調整

karateboy, 2021-08-11:
已經修正

haopingchung, 2021-08-13:
定值(0700-0730)
通訊中斷(0705-0710)
[image]

另外平台的"斷線"檢查時間應為定值檢查時間
[image]

karateboy, 2021-08-13:
了解....我頭昏了.

haopingchung, 2021-08-16:
- [ ] 1.定值還是顯示0700
[image]

另外定值檢查時間的設定應該是一個範圍?(0700~0730，檢查這30分鐘是否相同)

karateboy, 2021-08-16:
現在就是7:00  其實我建議定值檢查範圍長一點....比方說一小時. 因為我覺得定值常常是假警報

karateboy, 2021-08-16:
[image]

karateboy, 2021-08-16:
剛剛我設定改成7:30

haopingchung, 2021-08-17:
所以如果設定7:30 是檢查0700~0730嗎

closed 2021-08-19 by haopingchung