karateboy/SensorManager #66: 8/19：異常狀況定義

haopingchung, 2021-08-19:
定義(定時檢核機制)
定值：檢查範圍0700-0730，檢核時間0730(檢查範圍為檢核時間前30分鐘)
斷線：檢查時間0700-0730，檢核時間0730(有一筆數值就算正常)
充電：2000-2030，檢核時間2000-2030(有一筆非4就算異常)

- [ ] emai：寄信時間早上8點

"(日期)充電異常"改為"前一晚充電異常"

- [ ] 主畫面檢核資訊改為下圖

[image]

- [ ] 即時狀況

"紀錄時間"改為"檢核時間"

haopingchung, 2021-08-19:
[0819感測器會議.pptx](https://github.com/karateboy/SensorManager/files/7012453/0819.pptx)

haopingchung, 2021-08-24:
- [ ] 主畫面須調整
[image]

[image]

- [ ] 2.紀錄時間>檢核時間
[image]

karateboy, 2021-08-25:
[image]

karateboy, 2021-08-25:
[image]
已經更新