karateboy/SensorManager #98: 11/5：中研院DAS平台調整

haopingchung, 2021-11-05:
- [ ] 1. DAS新增登入畫面
- [ ] 2. admin及user的權限設定(user無法進入系統管理)
- [ ] 3. 主畫面右上角測值區域長度調整，超出範圍的測項透過下拉拉霸顯示
- [ ] 4. 儀器校正時的DO輸出，需可選擇DO輸出裝置(目前為自動選擇)

haopingchung, 2021-11-09:
- [ ] 1. 把使用者的管理者取消勾選後重新登入，主畫面趨勢圖就無法顯示、資料查詢也沒有測項可選擇
[image]
[image]

karateboy, 2021-11-09:
因為所屬群組沒有權限可以存取這些測項
剛剛已經把權限打開

haopingchung, 2021-11-10:
不過平台管理團隊也沒有開任何測項，還沒有相同問題
[image]

另外兩個群組中都有"管理員"選項可勾...設定使用者時又有"管理者"選項可勾，請問這兩個選項的區別是什麼?
[image]
[image]

karateboy, 2021-11-10:
查了一下程式碼, 是沒有作用. Orz

closed 2021-11-11 by haopingchung