karateboy/SensorManager #114: 12/3：中研院DAS新儀器斷線問題

haopingchung, 2021-12-03:
今日中研院測站新增一台TEOM(1405-F),已於conf資料夾中的AK protocol新增檔案

不過連上線後不久就會斷線(停用重啟後會正常一下),且在儀器狀態查詢那邊無法查詢到資料

再麻煩協助確認

[image]

karateboy, 2021-12-05:
看error log 讀到的字串是1405A221791210
並不是數字...感覺像是序號之類的
所以無法讀出讀值.

karateboy, 2021-12-05:
我把測項改成240 即時值  看看會不會改善

haopingchung, 2021-12-05:
連線7分鐘後斷線了...1405A221791210看起來是下命令1後會回傳的字串
(240回傳的東西很奇怪不像即時值,所以先用244小時值代替)
[image]
[image]

karateboy, 2021-12-07:
關掉蒐集儀器資訊後  暫時解決問題

haopingchung, 2021-12-20:
今日調整了1405-F的記事本，data跟input都只保留測項(245)，但還是會在30分的時候斷線
[image]

closed 2021-12-28 by haopingchung