karateboy/SensorManager #117: 12/16：中研院新測站DAS問題

haopingchung, 2021-12-16:
昨日中研院在雲科大的測站也安裝了DAS(1.3.22)，目前遇到以下幾個問題

- [ ] 1. page箭頭顯示異常
[image]

- [ ] 2.小時資料無法查詢(分鐘值正常)
[image]

- [ ] 3.趨勢圖跑不出來(會一直轉圈圈)
[image]

- [ ] 4.日報、月報無資料(因為沒有小時值?)

這台電腦當初在裝mongoDB(4.2)的時候,因為compass安裝一直跑不動,所以沒有安裝compass
遠端連線目前是透過chrome remote(wecc27548312@gmail.com)，teamviewer/anydesk都無法連線

karateboy, 2021-12-18:
chrome 桌面  必須登入chrome帳號  我試著登入時 google會要求傳送簡訊雙重認證

karateboy, 2021-12-20:
Thermail 43i的設定有誤, 導致啟動錯誤.
我手動把43i刪除後  就恢復正常了.
請將軟體更新到1.3.24版  謝謝

karateboy, 2022-01-07:
問題已找到, 當初為了讓使用者可以下載1個月的分鐘值趨勢圖
所以使用了MongoDB 4.4才有的功能. 所以在4.4之前的資料庫會造成不相容
在application.conf中現在增加一個選項below44 預設註解 (預設值是false)
在4.4之前的MongoDB就必須設定成true

my.mongodb{
	url = "mongodb://localhost"
	db = "logger2"
	# set true to be compatible with mongo 4.0/4.2
	below44 = true
}

haopingchung, 2022-01-09:
[image]

了解，目前剩下箭頭顯示的問題

karateboy, 2022-01-09:
這個是因為Windows 7 要安裝emoji 更新. 才能顯示這些新增加的圖型
我剛剛已經安裝了.

karateboy, 2022-01-09:
附帶一提  我在那台電腦上面安裝了小紅傘, 因為它可以把之前移除不乾淨的防毒軟體清掉.
(各家防毒軟體都有自動清除競爭對手軟體的能力)

haopingchung, 2022-01-10:
[image]
目前有顯示其中一組了，但另一組仍未顯示

karateboy, 2022-01-10:
這好像沒辦法  Windows 7只支援一部分的emoji. (這是unicode的圖形字形)  微軟後來也只更新到這為止

haopingchung, 2022-01-10:
或是換一個win7有支援的emoji呢

karateboy, 2022-01-11:
這我google很久  好像沒有甚麼特別好的方法....其實emoji是"字"型  這就像作業系統沒有這個字型  就顯示一個方框給你.
最好的解決方法好像只有升級到Win7 之後的作業系統

closed 2022-01-11 by haopingchung