karateboy/SensorManager #122: 12/29：picarro校正問題

haopingchung, 2021-12-29:
- [ ] 12/28 23:00 picarro的校正沒有成功啟動，29號10:48透過手動校正亦同(看log檔有error訊息)

haopingchung, 2022-01-03:
- [ ] 1/3發現picarro的測值在校正時沒有跟著變動，如下圖
[image]

karateboy, 2022-01-03:
現在是停用狀態?

haopingchung, 2022-01-03:
對 可以啟用

karateboy, 2022-01-08:
已經找到問題了  因為上升/持續/下降沒有設定

haopingchung, 2022-01-09:
透過DAS下校正之後有確實啟動了，不過還是有些奇怪的現象....
1、下校正時DAS端測值還是沒有變動(儀器端有)
2、從趨勢圖來看1/8 9:04開始測值就躺平了
[image]
3、1/9 4:50斷線，不過5:31、6:27、7:29、8:32、9:44、10:44都各有一筆測值

karateboy, 2022-01-09:
要不要再檢查一下儀器回報的順序  有沒有改變?

haopingchung, 2022-01-10:
是指回應的內容嗎?我看了4個測項，看起來是正常的

[image]
[image]

closed 2022-01-11 by haopingchung