karateboy/SensorManager #124: 1/4：中研院雲科大測站新增儀器

haopingchung, 2022-01-04:
1. 雲科大有一台資料蒐集器，裡面有氣象設備的數據，目前預計作法是像光譜儀一樣，建立網路磁碟機：
[image]，再解裡面的內容，檔案如下(副檔名為DAT)：
[CR800_Table_min.zip](https://github.com/karateboy/SensorManager/files/7806658/CR800_Table_min.zip)
[image]

karateboy, 2022-01-08:
有樣本檔案嗎?

haopingchung, 2022-01-09:
有啊 上面有個zip檔

karateboy, 2022-01-15:
昨天總算找出問題....但那台電腦無法一下子全部匯入.
所以分段  匯入...現在已經可以查到12月中的資料了
明天應該可以匯完

haopingchung, 2022-01-18:
今日重新看了下資料，發現有不少地方數據是斷掉的，最後一筆資料停在17號下午一點
[image]

karateboy, 2022-01-18:
看來是網路硬碟中斷造成的...剛剛點兩下網路硬碟又恢復了.
數值斷掉是合理的....看原始檔案  它有時資料是NaN 也就是沒資料.

haopingchung, 2022-01-18:
[image]
目前又沒有資料了@@

karateboy, 2022-01-19:
昨天我有把雲科大從頭開始讀那個檔案  所以大概要兩天才會到目前的時間....

karateboy, 2022-01-25:
雲科大的問題似乎是  排定系統自動執行時  Y:這種網路磁碟機會失敗
如果是開機後執行  就沒有這個問題

karateboy, 2022-01-27:
也許是忘記關檔案.  今天(1/27) 更新了. 再觀察看看

closed 2022-01-28 by haopingchung