karateboy/SensorManager #133: 1/28：中研院(雲科大)

haopingchung, 2022-01-28:
- [ ] 1、氣象資料蒐集斷線問題
- [ ] 2、"維修"狀態在一小時內比例需多少才會被標註(理論上超過25%就須標註)

karateboy, 2022-01-30:
氣象資料已經兩天正常了. 後來發現問題出在, 有時氣象資料會比其他測項先拿到資料. (其他儀器固定在30秒時計算更新前分鐘值)
其他儀器更新時, 資料庫語法是replace. 所以氣象資料就被蓋過去丟掉了. 
改成Upsert (Update or Insert)後就正常了

karateboy, 2022-01-30:
狀態問題已經找到, 當初在處理感測器時, 小時值計算採永遠有效.
現在已經恢復正規計算

karateboy, 2022-01-30:
1/25 重新計算後13:00, 小時值已經變為維修狀態

haopingchung, 2022-02-07:
忘記說了...還有emoji需要調整一下
[image]

haopingchung, 2022-02-07:
氣象資料查詢正常，但是下載excel後有時候會顯示無法開啟檔案
[image]

檔案名稱改掉後，就可正常開啟，是否是因為檔名過長導致

karateboy, 2022-02-13:
2/7 13:00人工註記重算小時值錯誤的bug 已經找到了. 已經更新

closed 2022-02-14 by haopingchung