karateboy/SensorManager #134: 1/28：鹽洲測站

haopingchung, 2022-01-28:
- [ ] 1、蒐集所有光化測站資料(新程式)
- [ ] 2、設定格式如下：
[image]

鹽洲測站teamviewer
ID：270904488
password：we27548312

karateboy, 2022-02-13:
Dear YS
 
屏東站統計軟體要試算的部分資料放在屏東站電腦桌面
 
資料夾名稱為 “忠明站Data”，再請協助軟體開發的部分
[image]

karateboy, 2022-02-13:
@haopingchung 這台電腦在哪? 看起來不是teamview的那台電腦. 桌面顏色也不同

haopingchung, 2022-02-13:
明日確認

haopingchung, 2022-02-14:
已請PE把檔案丟到網路磁碟機
鹽洲測站teamviewer
ID：270904488
password：we27548312
[image]

karateboy, 2022-02-18:
已初步完成.
目前放在鹽洲測站8000 port
測站設定檔案在application.conf
其他測站, 比照相同方式編輯設定
vocReader {
  enable = true
  monitors = [
    {id = "M1", name = "忠明站", lat = 0.0, lng = 0.0, path="\\\\localhost\\20220113_1"}
  ]
}

karateboy, 2022-02-20:
首頁還沒弄, 但我目前預計是打算等各站設定好後, 繪製地圖顯示狀態.
但VOC只有小時值, 測項又很多. 所以想討論一下.
另外VOC也沒有即時值, 所以即時值的頁面  我預計也會拿掉

haopingchung, 2022-02-21:
ys認為首頁放google地圖，然後地圖上顯示各站的位置就好

haopingchung, 2022-02-24:
地圖上忠明的位置不太顯眼，地圖上的icon可否放大一些

另外，忠明站的地點是：台中市南屯區公益路二段296號
試著在application.conf改經緯度後，地圖上的位置仍在基隆

karateboy, 2022-02-24:
已更新修正.

karateboy, 2022-02-27:
台東測站中控端已經上線

closed 2022-03-01 by haopingchung