karateboy/SensorManager #137: 1/28：10000系統debug

haopingchung, 2022-01-28:
- [ ] 1、各案場法規值獨立

- [ ] 2、警報紀錄須註明發生案場

- [ ] 3、按下測試所有案場都會灑水

- [ ] 4、法規值超標觸發灑水是否觸發到正確的案場

到時會在公司建立一組灑水模組，以便確認問題是否修正

haopingchung, 2022-03-14:
今日以test帳號測試(test@wecc.com.tw/test)，目前發現問題如下：

1、首頁無資料
[image]

2、查詢資料無法顯示
[image]

3、警報紀錄包含到其他案場
[image]

4、灑水須註明案場
[image]

5、有超標警告，但沒被觸發
[image]

karateboy, 2022-03-14:
都已修正

closed 2022-03-17 by haopingchung