karateboy/SensorManager #138: 2/14：中研院(台中、雲科)

haopingchung, 2022-02-14:
**雲科：**

- [ ] 1、氣象資料下載成excel後，會有無法開啟的問題
[image]

- [ ] 2、DAS更新完後，之後每日校正都不會啟動，要把sabio斷線重連後才會正常，是否每次更新後可以確認下sabio的連線狀態

**台中：**

- [ ] 1、fidas數據常有異常極高值或極低值的問題，以過去一周來說，PM1、PM2.5、PM10發生次數非常頻繁，PM4、PMtotal相對穩定
[image]
[image]
[image]
[image]
[image]

haopingchung, 2022-02-15:
- [ ] 1、小時值的計算方式如下：
無論大於或小於75%，計算小時值時皆只計算有效數據，若整小時都為無效，則顯示"-"
狀態碼則維持原規則(若<75%，取最多狀態為狀態)

karateboy, 2022-02-15:
1.  雲科檔案名稱問題已經修正
2. 台中fidas出現0值的原因是通訊出現問題, 當一分鐘都沒有值的時候, 會無法取平均. 這邊資料處理有錯誤. 會產出0的數值. 正確做法應該是沒有數值.
3.

haopingchung, 2022-02-17:
2/17觀察fidas 近24小時的值，目前PM1仍有類似問題，測值會掉到-2500左右，其餘測項看起來有比較穩定
[image]

closed 2022-02-17 by haopingchung