karateboy/SensorManager #139: 2/16：中研院新功能需求

haopingchung, 2022-02-16:
- [ ] 1、台中及雲科的測站的thermo CO，儀器每小時會自動做零點(0分~12分)，因此希望有一個自動標註的功能，自動將每小時前12分鐘的數據標註成無效(暫定為維修)，以免小時值的數據被拉低

karateboy, 2022-03-05:
fixed 
新增autoState於application.conf
參數意義如下
instID (設備ID)
state (狀態碼)
period (時間, 目前只有Hour)
time (觸發分鐘)
autoState=[
	{ instID="48i"
		state="031"
		period="Hour"
		time="0"
	},
	{ instID="48i"
		state="010"
		period="Hour"
		time="13"
	},
]

haopingchung, 2022-03-07:
大致上沒有問題

不過會與校正衝突到(2300~2340再執行校正)

是否可以加入除外時間，例如2300~2400不啟用自動標註狀態功能
[image]

haopingchung, 2022-03-11:
跟中研院討論後，目前是打算直接調整自動校正時間...，等復電後會再確認

karateboy, 2022-03-11:
如果  需要把功能關掉, 
再告訴我.  把application.conf 中變成註解即可
autoState=[
#	{ instID="48i"
#		state="031"
#		period="Hour"
#		time="0"
#	},
#	{ instID="48i"
#		state="010"
#		period="Hour"
#		time="13"
#	},
]

closed 2022-03-14 by haopingchung