karateboy/SensorManager #141: 2/18：中研院

haopingchung, 2022-02-17:
雲科：

- [ ] 1、DAS首頁左上方的英文名錯誤(UAPRS應改為APCRS)
- [ ] 2、由於改回35，因此下次更新時需再調整一次氣象資料名稱
- [ ] 3、今日將16號1350~17號1900的分鐘資料從**維修>有效**後重算小時值，但有不少小時值仍呈現維修
- [ ] 4、 36版似乎有狀態跳動的問題，我會再安排時間確認詳細情況
- [ ] 5、雲科das若有更新，可先將檔案放在電腦中，我會找時間測試

台中：

- [ ] 1、觀察fidas近24小時的的分鐘值，PM2.5、10在2/17 15:28及16:51同時有異常值，PM1次數則相對較多
[image]
[image]
[image]

karateboy, 2022-02-21:
中研院台中37版本已經放在資料夾中. 請測試後確認.

karateboy, 2022-02-21:
雲科37版也已經放在資料夾中, 請驗證後上板

haopingchung, 2022-02-24:
目前台中是38版

- [ ] 最近24小時PM1仍有一些奇怪的測值，其餘項目則沒有發生了

[image]

- [ ] 小時值計算，以excel來說2300時應計算2300、2342~2359的平均值，但實際計算是2341~2359(多算了校正恢復的部分)
[220223SO2NOY.xlsx](https://github.com/karateboy/SensorManager/files/8130696/220223SO2NOY.xlsx)

karateboy, 2022-02-25:
奇怪...我下載分鐘值得到的資料和你不同. 14.06標註成正常?
[image]

karateboy, 2022-02-25:
Excel狀態碼問題已修正

haopingchung, 2022-02-25:
雲科大

- [ ] 1、左上角仍是UAPRS
[image]

- [ ] 2、重新計算17號的小時值後，資料就不見了，我在分鐘值標註的是"有效數據"(黃色)，所以在小時值計算時也應會納入計算?
[image]

karateboy, 2022-02-25:
@haopingchung 雲科大 程式我放在那邊沒有啟動. 會不會是執行到原本的版本? 我等下更新啟動好了

karateboy, 2022-02-25:
雲科title 字樣已經更新.
重算"人工註記有效"的小時, 已經會變成"正常"小時

haopingchung, 2022-02-25:
目前還有零星小時值，似乎未被計算到
[image]

karateboy, 2022-02-26:
找到問題了, 已經更新到雲科大

closed 2022-03-01 by haopingchung