karateboy/SensorManager #146: 3/17：灑水系統debug

haopingchung, 2022-03-17:
- [ ] 1、儀錶板地圖上無顯示濃度與座標

- [ ] 2、儀錶板即時值顯示"-"，但資料查詢有
[image]
[image]
可能與感測器丟資料的時間有關，是否有辦法處理

- [ ] 3、觸發ADAM的位址調整

接線在RL1，平台設定在0卻會觸發成功
另外測試按鈕的觸發位址與濃度觸發應相同

- [ ] 4、設定名稱可否修改，避免錯亂
[image]

karateboy, 2022-03-19:
儀錶板顯示的沒有資料, 但後來查詢有. 主要是反映這個狀況.
1. 假設目前查詢最新資料是10:05分, 感測器A,B,C,D, 其中C還沒有10:05分的資料. 所以C是-
2. 到了10:06分, C的10:05資料來了. 所以之後去查詢. C的資料就出現了.

karateboy, 2022-03-19:
DI/Relay已經透過  儀器管理調整

karateboy, 2022-03-20:
59.124.12.65 無法更新軟體. Teamview/anydesk一傳檔案就會斷線, 請幫助檢查一下

haopingchung, 2022-03-21:
- [ ] 1、目前在儀器管理 只有看到relay的調整 沒看到DI的調整

[image]

- [ ] 2、另外第三點有關平台觸發的RELAY位置(16?17?)，這部分有做調整嗎

karateboy, 2022-03-21:
@haopingchung  relay 是跟DI同樣的編號. 我不覺得要分開設定. 16, 17已經改了  你可以測試看看. 0 => 16, 1=>16

haopingchung, 2022-03-30:
今日我使用test帳號測試，結果如下：

接線：RELAY:3+/3-，DI:3/GND

按下測試，灑水系統無動作(使用蜂鳴檔測試其餘RL也都沒有導通)

透過法規值觸發，結果正常

karateboy, 2022-03-30:
已修正測試按鈕bug.

haopingchung, 2022-04-01:
今日測試了幾次，測試按鈕也已經正常
不過偶爾會有觸發灑水後沒有切回standby的情況，有什麼機制可以防止呢

https://user-images.githubusercontent.com/84848815/161180604-dcf1794b-2794-4b85-b085-b294e44b19f8.mp4