karateboy/SensorManager #150: 4/7：資料蒐集系統

haopingchung, 2022-04-08:
預計4月底前先完成下述1~4項，並執行功能測試

- [ ] 1、完成搭配sql server的資料蒐集器
- [ ] 2、完成搭配sql server的中控端
- [ ] 3、database格式：一分鐘一筆資料
- [ ] 4、新增手動輸入MB值功能

- [ ] 5、LINE告警功能
- [ ] 6、timer停止計時問題排除
      (1).校正開始時就記錄span&standby執行時間
      (2).程式啟動時，自動對所有分析儀及校正器下standby命令
      (3).standby命令於校正結束後一分鐘再執行一次

- [ ] 7、由中控端上傳資料至環保署，需包含以下功能：
      (1)自動上傳功能
      (2)手動上傳功能
      (3)測試功能
      (4)異常數據過濾不上傳功能(最大最小值)
      (5)上傳失敗時告警功能(line or email)
      參考畫面如下
[image]

karateboy, 2022-05-16:
新北市的資料蒐集器已經是SQL server版本, 可以協助測試驗證前面幾項功能
在application.conf 新增logger區設定, db=sql 就可以使用sql server作為資料庫,
設定nosql 就可以回到原本mongodb
logger {
  storeSecondData = false
  selfMonitor = true
  trendShowActual = true
  **db = sql**
}

karateboy, 2022-06-04:
已增加手動輸入M/B值功能.
規則如下, 
1 設定校正
2. 如果有M/B值, 會優先以此校正
3. 若沒有M/B值, 會以最近校正紀錄校正

closed 2022-06-13 by haopingchung