karateboy/SensorManager #151: 4/7：感測器平台調整

haopingchung, 2022-04-08:
- [ ] 1、將7000移至65
- [ ] 2、開放調整首頁地圖要顯示的測項，並在地圖外標註
- [ ] 3、目前灑水系統登入時地圖會看到整個台灣，建議根據感測器位置調整大小
- [ ] 4、灑水系統設定簡化(我還在思考怎麼規劃，過幾天再提出)

haopingchung, 2022-04-13:
目前有將GS系列新增到Y1LO01群組中，不過下載excel時並沒有在清單中
[image]
[image]

karateboy, 2022-04-13:
我把那幾台勾選入Y1LO01群組...就可以下載了
[image]

karateboy, 2022-04-13:
我發現當初寫這這樣: 如果沒有選擇群組, 沒有代碼的感測器就不會輸出到Excel. 這規則不知道當初是為什麼設定的. 你直接輸出感測器, 看不到這些GS開頭的感測器, 就是因為他沒代碼. (如果有選定群組, 就會輸出)

haopingchung, 2022-04-14:
有關灑水系統設定畫面整合的部分，目前想法如下：
1、測點管理及感測器管理合併，保留感測器代碼、主題、名稱、測項
2、設定帳號時直接選擇要看的感測器，而不是選擇群組

目前群組的功能只有把感測器分群?似乎有點多餘

karateboy, 2022-04-26:
灑水系統地圖實際上已經根據感測器位置  將地圖中心放在感測器的範圍中心. 但因為地圖Zoom level太小, 所以感覺感測器集中在一點

closed 2022-06-27 by haopingchung