karateboy/SensorManager #153: 4/14：7000平台檢核功能問題

haopingchung, 2022-04-14:
- [ ] 1、7000平台的斷線檢查似乎故障了，有部分感測器斷線，14日早上卻沒有被偵測到
[image]

karateboy, 2022-04-25:
台積電那台電腦越來越奇怪....軟體完全沒動, 小時值卻不會更新. 連log也沒有
剛剛重啟後, 又恢復正常

karateboy, 2022-05-31:
看Log每天7:29 44左右都會執行檢測
2022-05-29 07:29:44,544 [INFO] from application in application-akka.actor.default-dispatcher-12843 - **update daily error report 2022-05-29T00:00:00.000+08:00**
2022-05-30 07:29:43,798 [INFO] from application in application-akka.actor.default-dispatcher-14020 - **update daily error report 2022-05-30T00:00:00.000+08:00**
2022-05-31 07:29:42,976 [INFO] from application in application-akka.actor.default-dispatcher-14596 - **update daily error report 2022-05-31T00:00:00.000+08:00**

haopingchung, 2022-06-02:
所以是有執行,但沒有檢測到嗎@@?

karateboy, 2022-06-05:
終於找到問題了, 原本測站的資料表不知道為啥不見了. 需要重新匯入...
你有備份嗎?

karateboy, 2022-06-05:
終於找到問題了  MongoDB 再查詢時因為資料量太大, 超出記憶體的限制. 
允許MongoDB使用磁碟機當暫存  就解決了

haopingchung, 2022-06-06:
ok謝謝，我會再觀察一兩天

closed 2022-06-07 by haopingchung