karateboy/SensorManager #156: 6/7：65平台新增異常檢核

haopingchung, 2022-06-07:
65平台目前也需要新增異常檢測功能，目前規劃內容與7000差不多，詳細說明如ppt
[城鄉計畫感測器email告警.pptx](https://github.com/karateboy/SensorManager/files/8852080/email.pptx)

karateboy, 2022-06-22:
功能已經啟用, 使用方式為到使用者管理.
更新使用者 "通報email"
有設定的使用者, 每天早上會收到自己群組感測器的檢測email

haopingchung, 2022-06-23:
今日早上有收到桃園群組email，不過斷線的數量太多(43台)，桃園群組總共只有10台而已
[image]
[image]

haopingchung, 2022-06-27:
目前每天收到的mail桃園都是10台斷線，不過實際上應該只有352818664021150斷線才對
[image]

karateboy, 2022-06-27:
前天有發現  已經修正. 但因為Java更新, 又造成安全性的問題  email寄不出來. 明天早上應該就可以看到結果

karateboy, 2022-06-29:
你也可以試看看, 兩個email, 用分號隔開填入. 應該也是可以收到.

haopingchung, 2022-06-30:
- [ ] 1、今日雲林沒有收到異常檢核email，不知道是不是因為我填了兩個信箱的原因?
[image]

- [ ] 2、定值檢核目前是檢核哪一個測項呢?如果是PM2.5的話，今日早上桃園有兩台定值，不過沒有被偵測到
[image]
[image]

- [ ] 3、使用者可否自訂要被檢核的測項(定值部分)

karateboy, 2022-06-30:
1. 看log似乎是有一個沒有位置的感測器, 在產出google map的連結時, 發生錯誤無法把信寄出.
2. 之前沒有啟用定值的檢查, 今天已經啟用. 明天應該就可以看到結果

karateboy, 2022-06-30:
3. 可以, 但使用者介面會不會太繁瑣? 目前定值是只有檢測PM2.5. 當檢測期間PM2.5的最大值和最小值相等, 就判定是定值

haopingchung, 2022-07-01:
1、城鄉的感測器告警好像不包含google maps的連結@@?
3、城鄉的感測器主要是看H2S跟NH3，所以目前只要新增這兩個測項的檢查就ok~

karateboy, 2022-07-01:
感測器網站的位置有特別匯入，城鄉的感測器位置沒有匯入，所以沒有座標。一個解決方式，就是使用感測器回報的經緯度.

haopingchung, 2022-07-04:
那再麻煩你先這樣處理看看

karateboy, 2022-07-05:
更新了  明天早上看一下定值的通知信  應該會有NH3, H2S

closed 2022-07-06 by haopingchung