karateboy/SensorManager #158: 6/13：新北測站DAS

haopingchung, 2022-06-13:
**1~7項預計6/19前完成**
- [ ] 1、在主畫面新增CDX上傳狀態(包含最近上傳時間、上傳失敗紀錄)
- [ ] 2、在"CDX上傳設定"處，新增上傳紀錄歷史資料
- [ ] 3、在儀器管理處若有啟用設備，主畫面不應發生測項消失情形
- [ ] 4、主畫面測項新增流水號
- [ ] 5、6/13 11:12 CO斷線，再請查明原因
- [ ] 6、每小時檢查資料庫，若前一小時分鐘資料<45筆，則自動停用重啟該設備
- [ ] 7、目前中控端的資料檢核是以分鐘為單位，是否可改為小時
------------------------------------------------------------------------------------------------------------------------------------
- [ ] 8、校正結果與分鐘記錄不同，先透過每天校時與timer重置來測試看看問題是否可以解決

karateboy, 2022-06-14:
5. CO斷線是由於連線被T300中斷, 所以下達命令時就發生錯誤. 由於之前都是假設讀取命令結果發生錯誤, 所以沒有處理到這個狀況

karateboy, 2022-06-14:
4. 流水號已經增加上去

karateboy, 2022-06-18:
3. 已經修正. 主因是SQL database race condiction.

karateboy, 2022-06-18:
7. 中控端除非設定稽核分鐘值, 否則都只會稽核小時值. 警報紀錄中連續的小時值稽核, 是因為那些小時值都在同時上傳. (例如斷線一陣子後, 監測站連線恢復, 重新上船好幾個小時的資料. 造成同時被稽核. 警報記錄訊息會改進包含稽核的小時資訊

karateboy, 2022-06-18:
6. 已經實作, 可將儀器斷線. 看是否於整點重啟. (Log會顯示)

haopingchung, 2022-06-22:
- [ ] 1、主畫面測項將拉霸往下滑後會有顯示問題
[image]

- [ ] 2、CDX上傳紀錄可否移到最前面
[image]

haopingchung, 2022-06-27:
- [ ] CDX上傳的最大及最小值沒有生效問題修正

karateboy, 2022-06-27:
前幾天已經生效了
[image]

karateboy, 2022-06-27:
T700 重啟後, 會先執行SEQ 99 (Standby 指令)

closed 2022-06-30 by haopingchung