karateboy/SensorManager #162: 6/27：65感測器數據issue

haopingchung, 2022-06-27:
- [ ] 1、最近發現感測器網站在每晚12點左右，都會有幾分鐘的缺值，使用MQTTBOX監控後發現，除23:59外，00:00、00:01有收到資料，但沒有記錄進資料庫，如圖所示

[image]
[image]
[image]

haopingchung, 2022-06-27:
感測器平台(7000)有數據更新需求如下：
1、imei"352818662097327"須改為"355001090067132"
2、屏東46台數據需要更新，提供csv檔如下
https://drive.google.com/file/d/1N_OVEJu4PqhQs2SYyWwQMwL5c3oG-eoM/view?usp=sharing

karateboy, 2022-06-29:
感測器最近常常傳回一些和先前格式不同的資料. 也許是這個緣故

haopingchung, 2022-06-30:
好像是所有感測器都會遇到這個狀況，我今晚再觀察另一台看看

haopingchung, 2022-07-04:
- [ ] 7000感測器網站設備顯示名稱更改

closed 2022-07-06 by haopingchung