karateboy/SensorManager #165: 7/6：感測器issues

haopingchung, 2022-07-06:
**7000**
- [ ] 1、imei"352818662097327"改為"355001090067132"
- [ ] 2、屏東46台數據需要更新，提供csv檔如下
https://drive.google.com/file/d/1N_OVEJu4PqhQs2SYyWwQMwL5c3oG-eoM/view?usp=sharing

- [ ] 3、查詢感測器資料時的顯示名稱，希望自定義，方便快速辨識感測器(如下兩張圖)
[image]
[image]

**65**
- [ ] 1、目前首頁地圖的顯示濃度定義不清楚，希望可開放使用者調整
(1)、要顯示在地圖的測項
(2)、在地圖外備註目前顯示的測項為何
[image]

karateboy, 2022-07-07:
2. H2S, NH3 定值通報今天已經送出.

haopingchung, 2022-07-07:
> 2. H2S, NH3 定值通報今天已經送出.

有看到了~不過雲林的email仍未收到

karateboy, 2022-07-15:
1. 匯入感測器Excel問題已經解決.
2. 改imei那台感測器, 改不回來了. 因為在先前轉換的過程中, 因為程式有bug. 所以還沒寫入到資料庫. 舊的資料就被我刪掉了. 解決方法可否從廣域那邊拿到資料, 然後用匯入的?

haopingchung, 2022-07-21:
> 2. 改imei那台感測器, 改不回來了. 因為在先前轉換的過程中, 因為程式有bug. 所以還沒寫入到資料庫. 舊的資料就被我刪掉了. 解決方法可否從廣域那邊拿到資料, 然後用匯入的?

剛才我看了下這兩台的數據 看起來都還有在平台上
[image]

karateboy, 2022-07-29:
65清除資料的程式找到了, 會每天清除60天之前的分鐘值. 目前已經移除

closed 2022-08-01 by haopingchung