karateboy/SensorManager #167: 7/7：新竹消防局DAS(新案子)

haopingchung, 2022-07-07:
- [ ] 1、新竹消防局需要一套DAS系統，現場儀器為CR300，目前有範例檔案如下：(原為.dat檔，因上傳限制我改為.csv檔)
[CR300Series_Test.csv](https://github.com/karateboy/SensorManager/files/9062805/CR300Series_Test.csv)
[CR300資料說明.xls](https://github.com/karateboy/SensorManager/files/9062806/CR300.xls)

- [ ] 2、待資料蒐集完成後，首頁畫面需要更改(到時候會再說明如何調整)

karateboy, 2022-07-09:
已經驗證可以用原本的CR800氣象資料讀取, 設定檔application.conf增加參數指定model
若無指定, 預設是CR800

weatherReader {
  enable = true
  # model CR800/CR300
  **model = CR300**
  # dir = "Y:/CR800_Table_min.dat"
  dir = "Y:/CR300Series_Test.dat"
}

karateboy, 2022-07-09:
消防局的anydesk又連不上了. 等禮拜一有人在安裝處理. 另外我懷疑消防局的anydesk是不是免安裝版?
會不會重開機後就不見了

haopingchung, 2022-07-11:
連線已恢復，anydesk的問題我已經處理好了，以後應該就可以正常使用

karateboy, 2022-07-11:
目前首頁是可選擇顯示風速/風向/雨量最近一小時趨勢圖, 還有測項玫瑰圖
之後可討論在客製化顯示那些項目. 由於沒有其他儀器, 所以即時趨勢圖和數據不會顯示

haopingchung, 2022-07-18:
- [ ] 上週五跟ys討論後，認為畫面部分需要做些修改，如ppt
[氣象局畫面調整拷貝.pptx](https://github.com/karateboy/SensorManager/files/9129073/default.pptx)

karateboy, 2022-07-23:
1. 及時狀態已更新 (風標, 風速, 陣風, ) 每3秒更新一次. 風瑰圖1分鐘更新一次
2. 最大風速小時計算及15分鐘計算統計以最大值計算

closed 2022-07-25 by haopingchung