karateboy/SensorManager #173: 8/4：資料蒐集系統bug

haopingchung, 2022-08-04:
- [ ] 1、日報表計算average、max、min、有效率時，須排除無效數據
[image]

- [ ] 2、月份時報表的每行每列的平均值似乎有異常，看起來是因為直接除以31導致的
[image]

- [ ] 3、LOG內容簡化、並加入儀器名稱

- [ ] 4、雲科大雨量小時值重新計算，由"每小時最大值"改為"每小時加總"

- [ ] 5、每日自動校正log紀錄確認，以雲科大為例
[image]

karateboy, 2022-08-07:
相關錯誤已經修正. Log 移除thread的資訊, 簡化顯示.
至於各儀器log 顯示, 修改工程比較大. 建議改以不同工作項目紀錄

haopingchung, 2022-08-08:
- [ ] 1、重算雲科大小時值，計算完成後發現被計算區間的小時值沒有資料

karateboy, 2022-08-08:
雲科大重算後小時值已經有資料

closed 2022-08-10 by haopingchung