karateboy/SensorManager #179: 10/08：雲科大DAS自動狀態問題

haopingchung, 2022-10-08:
- [ ] 1、雲科大有設定CO每小時的0~12分自動標註為維修，但今日發現10/5 1800後CO所有分鐘值都是維修狀態而沒有自動切回正常，剛才已經手動將數據先註記並重啟DAS，之前新北三峽有遇過類似問題，再麻煩確認是否有將當時的修正套入雲科，謝謝。

karateboy, 2022-10-08:
從Log紀錄, 18, 19...之後每個小時都有切換狀態....
18點開始有問題是因為有人手動去把48i (CO)的儀器停用然後又啟用   
造成系統同時有兩個子程式同時收CO的資料....結果只有其中一個收到切換正常的指令...
可能要先問一下是誰當時做了這個操作. 

**2022-10-05 18:12:27,879 [INFO] application - Stop collecting instrument 48i 
2022-10-05 18:12:27,889 [INFO] application - remove List(CO)
2022-10-05 18:12:30,718 [INFO] application - TcpModbus.Thermo.48i_8 is created.
2022-10-05 18:12:31,770 [INFO] application - TcpModbus.Thermo.48i_9 is created.**
2022-10-05 18:12:31,972 [INFO] application - Actor[akka://application/user/dataCollectManager/TcpModbus.Thermo.48i_8#855300499]: connect 60.222.121.48
2022-10-05 18:12:31,974 [INFO] application - Actor[akka://application/user/dataCollectManager/TcpModbus.Thermo.48i_9#-821635899]: connect 60.222.121.48
2022-10-05 18:12:32,161 [INFO] application - Probing supported modbus registers...
2022-10-05 18:12:32,161 [INFO] application - Probing supported modbus registers...
2022-10-05 18:12:35,059 [INFO] application - Finish probing.
2022-10-05 18:12:35,059 [INFO] application - Finish probing.
2022-10-05 18:12:35,160 [INFO] application - Probing supported modbus registers...
2022-10-05 18:12:35,160 [INFO] application - Probing supported modbus registers...
2022-10-05 18:12:37,591 [INFO] application - Finish probing.
2022-10-05 18:12:37,665 [INFO] application - Finish probing.
2022-10-05 18:13:00,405 [INFO] application - AutoState=>AutoStateConfig(48i,010,Hour,13)
2022-10-05 18:13:00,405 [INFO] application - SetState(48i, 010)
2022-10-05 18:13:00,430 [INFO] application - Actor[akka://application/user/dataCollectManager/TcpModbus.Thermo.48i_9#-821635899] => 正常
2022-10-05 18:18:17,595 [ERROR] application - Read timeout!

karateboy, 2022-10-08:
下一版會針對這種race condiction作保護....其實停用啟用...應該只能有一個蒐集子程式起來. 使用者可能是快速按鈕..就跑出兩個...

closed 2022-10-24 by haopingchung